MicroBahner/MobaTools #2: Servo Speedcontrol not working after detaching/attaching Servo

antonsperling, 2019-12-23:
When a Servo is attached only once and not detached during program, speed control works really well.
But when this Servo is detached (using detach()) and then attached again (using attach(pin)), calling setSpeed() does not work anymore. Servo moves immeadiatly to target position using write(angle).

MicroBahner, 2019-12-23:
Hi Anton,
does the setSpeed() command really not work anymore? Or is only the first write() after reattaching with full speed?
When detaching the servo, all information about it is lost. It's just the same as before the first attach(). In this situation the actual position of the servo is not known. You could reattach to another pin with another servo. Controlling the speed is only possible if  the actual position of the servo is known. That's the reason why the first write() after an attach() is always full speed.
If you know that you reattach to the same servo, the first write after reattaching should be to the same position as the last write bevor detaching.
After this first write, all following should execute with the set speed. But you must set the speed again after reattaching
Franz-Peter

antonsperling, 2019-12-28:
Hey Franz-Peter,
thanks for this advise, and sorry for the late replay. It was Weihnachten in Kölle ;-)
Tried it with restoring the last position before servo was detached. But that is not all. I had to add a delay before next position is written (just some ms), otherwise the servo again will move at full speed. Maybe thats the time the servo itself needs to recognize its own position.
Here is my quick&dirty sample code:
`#include <MobaTools.h>
#define Servo Servo8

const int SERVO_PIN = 3; // pin number on UNO
Servo servo;

void setup() {

  Serial.begin(115200); // Open serial monitor at 9600 baud to see ping results.

  servo.attach(SERVO_PIN);
  servo.setSpeed(20, HIGH);
  Serial.println(F("Moving to 0 degree"));
  servo.write(0);
  delay(4000);

  servo.detach();
  servo.attach(SERVO_PIN);
  Serial.println(F("Restore last position on 0 degree after first reattach. Should not move."));
  servo.write(0);
  servo.setSpeed(20, HIGH);

  delay(3000);

  Serial.println(F("Moving to 90 degree after first reattach. Moves slowly."));
  servo.write(90);
  delay(4000);

  servo.detach();
  servo.attach(SERVO_PIN);
  Serial.println(F("Restore last position on 90 degree after second reattach. Should not move."));
  servo.write(90);
  servo.setSpeed(20, HIGH);
  delay(30); // ATTENTION: By adding this delay, the servo moves at desired speed....
  Serial.println(F("Moving to 180 degree after second reattach. Moves at full servo speed!!!"));
  servo.write(180);
  delay(4000);

  servo.detach();
  servo.attach(SERVO_PIN);
  Serial.println(F("Restore last position on 180 degree after third reattach. Should not move."));
  servo.write(180);
  servo.setSpeed(20, HIGH);
  delay(30); // ATTENTION: By adding this delay, the servo moves at desired speed.
  Serial.println(F("Moving to 0 degree after third reattach. Moves at full servo speed!!!"));
  servo.write(0);
  delay(4000);

}

void loop() {
}`

MicroBahner, 2019-12-29:
Hi Anton,
ich schreib jetzt einfach mal deutsch ( Kölsch kann ich nicht ;) )
Du hast tatsächlich noch einen timing-Fehler gefunden. Das Problem entsteht, wenn detach, attach, 1. und 2. write zu schnell hintereinander kommen. Die Pulse für den Servo werden ja nur alle 20ms ausgegeben. Passiert das nun alles innerhalb der 20ms und zwischen 1. und 2. write wird noch kein Impuls erzeugt, so wird auch der 2. write noch als 'fist write' interpretiert. Deshalb war der 30ms delay notwendig - da wurde dann vor dem 2. write ein Impus erzeugt.
Ich habe das bereinigt und auch schon hochgeladen. Es ist allerdings noch nicht als neue Version freigegeben. Wenn Du es mal testen willst, must Du es als .zip hier herunterladen.

antonsperling, 2019-12-29:
Hi FP,
habe die neue Version mit meinem Beispiel Code getestet. Sieht gut aus:-) Danke!

closed 2019-12-29 by antonsperling

MicroBahner, 2019-12-29:
Super, ist jetzt auch als 1.1.5 freigegeben und über den Bibliotheksverwalter isntallierbar.