MicroBahner/MobaTools #4: Variable Speed Control for stepper motor

gxdeange, 2020-08-25:
Hi MicroBahner,

Would you kindly advise if this library has capability for variable speed based on analog input and then mapping to a RPM or step/sec range?

I do this with mwc_stepper for AVR but would like to do the same for STM32F103C controller with a stepper library.

Cheers
George

MicroBahner, 2020-08-26:
Hi George,
there is no function to do this automatically. But of course you can read your analog input , map it to the desired speed, and then set this speed with the setSpeed()  (rpm) or setSpeedSteps()  (steps/sec) methods. This can be done while the motor is moving and it will adjust its speed accordingly.

Unlike most other stepper libs there is no 'run' method to create the steps. The steps are generated in the background by interrupt, and with setSpeed() or setSpeedSteps() you can adjust the step generation at any time. What you have to keep in mind is that the motor movement is independent from the sketch. When the motor has been started, the sketch continues to run while the motor is rotating. You can do almost anything without it affecting the movement - even during delay() the motor continues to run (to the target position). But you can also use the methods of the lib to control the movement (e.g. speed) at any time.

Regards
Franz-Peter

gxdeange, 2020-08-26:
Thanks Franz-Peter,

That is exactly what I was hoping for.

Cheers and have a great day.
George

closed 2020-08-28 by gxdeange