MicroBahner/MobaTools #5: RampLen() Does not seem to work

gxdeange, 2020-09-09:
Hi MicroBahner,

I am having an issue with getting the RampLem() to work and am not sure if it is the method or where I have placed it within my code.

I have tried to place in different parts of the code but I assume it needs to go within the void(setup)? I have also tried it within the void loop() itself which seemed to work but the examples do not show it within the void loop().

Regards and have a great day
George

MicroBahner, 2020-09-10:
Hi George,
Can you show an example where it does not work?
You can call setRampLen() at any time and it should be active the next time the speed of the stepper changes. Even if the stepper is inside a ramp, you can change the length and it should take effect immediately.

Regards
Franz-Peter

gxdeange, 2020-09-10:
Hi Franz-Peter, Sketch below
// MobaTools library installed from https://github.com/MicroBahner/MobaTools

// RBD_Timer Copyright (c) 2015 Alex Taujenis - MIT License
// https://github.com/alextaujenis/RBD_Timer

#include<MobaTools.h>
#include <RBD_Timer.h>

#define DIR 10               // Direction Pin Output
#define STEP 9              // Step Pin Output
#define EN_SW 5             // Enable Input
#define EN_LED 7            // LED Output +ve Trig
#define EN 8                // Driver Enable Output
#define FAN 6               // FAN Output

#define DIR_IN A2           // Pin A0 Direction Input
#define SPEED1 A3           // Pin A2 Speed1 Input
#define SPEED2 A4           // Pin A3 Speed2 Input

#define Limit01 3           // Pin 2 connected to Limit switch out
#define Limit02 4           // Pin 3 connected to Limit switch out

int SPD;
int s1;
int s2;
int dir_in;
int en_out;
int Lim1;
int Lim2;

MoToStepper stepper(800, STEPDIR); // 1/8 microstep setting, 3 wire driver board

RBD::Timer timer;

long timing = 0;
int stat = LOW;       //Sets the intial enable state to High (disabled for 4988 & 8825. Can be set to LOW for TB67S2xx)
int reading;          //The current reading from the input pin
int previous = HIGH;  //Sets previous reading to Low (set to high for TB67S2xx)
long debounce = 200;  //Debounce time

void setup() {

  stepper.attach(STEP, DIR);
  stepper.setSpeed(0);
  stepper.setRampLen(50);

  timer.setTimeout(5000);
  timer.restart();

  pinMode(DIR, OUTPUT);
  pinMode(STEP, OUTPUT);
  pinMode(EN_SW, INPUT_PULLUP);
  pinMode(EN_LED, OUTPUT);
  pinMode(EN, OUTPUT);
  pinMode(FAN, OUTPUT);

  pinMode(Limit01, INPUT_PULLUP);   //For N/C limit switch
  pinMode(Limit01, INPUT_PULLUP);   //For N/C limit switch

}

void loop() {

  dir_in = analogRead(DIR_IN);
  s1 = analogRead(SPEED1);
  s2 = analogRead(SPEED2);

  reading = digitalRead(EN_SW);
  if (reading == LOW && previous == HIGH && millis() - timing > debounce) { // This is for earth triggering switch
    if (stat == HIGH)
      stat = LOW;
    else
      stat = HIGH;

    timing = millis();
    timer.restart();
  }

  digitalWrite(EN, stat);         // Enable Output
  digitalWrite(EN_LED, stat);     // Enable LED Output

  if (stat == LOW) {
    if (timer.onRestart()) {

      digitalWrite(FAN, LOW);     // Fan off
    }
  }

  if (stat == HIGH) {

    digitalWrite(FAN, HIGH);      // Fan on
  }

  previous = reading;

  if ((dir_in) > 602) {           // Check DIR value
    Lim1 = digitalRead(Limit01);  // Read limit switch value

    if (Lim1 == HIGH) {

      stepper.rotate(0);          // Stop if limit switch open
      stepper.setZero();          // Reset internal counter

    }

    else {

      SPD = map(s1, 0, 1023, 10, 1850);  // Map SPD_IN value to SPD [suited to SPET_REV of 1600]

      //stepper.setRampLen(100);
      stepper.setSpeed(SPD); // Speed variable passed to setSpeed()
      stepper.rotate(1);          // Run motor (direction 1)

    }
  }

  if ((dir_in) < 422) {           // Check DIR value

    Lim2 = digitalRead(Limit02);  // Read limit switch value

    if (Lim2 == HIGH) {

      stepper.rotate(0);          // Stop if limit switch open
      stepper.setZero();          // Reset internal counter

    }

    else {

      SPD = map(s2, 0, 1023, 10, 1850);  // Map SPD_IN value to SPD [suit to STEP_REV fo 1600]

      //stepper.setRampLen(100);
      stepper.setSpeed(SPD);         // Speed variable passed to setSpeed()
      stepper.rotate(-1);                 // Run motor (direction -1)

    }
  }

  if (((dir_in) <= 602) && ((dir_in) >= 422)) { // Check DIR value

    stepper.rotate(0);                          // Stop motor
    stepper.setZero();                          // Reset internal counter
  }
}

gxdeange, 2020-09-10:
I also use a different timer library and I quite haven't worked out your timer library

gxdeange, 2020-09-10:
You will notice that I have commented out a few RampLen() within the void(loop).

I found it kind of worked but caused the speed to behave abnormally

MicroBahner, 2020-09-10:
Thanks, I will have a look at it.

gxdeange, 2020-09-10:
Cheers. Have a great day

gxdeange, 2020-09-11:
Hi Franz-Peter,

I managed to work out what I was doing wrong. RampLen works as expected.

Cheers
George

gxdeange, 2020-09-11:
Hi Franz-Peter,

Apologies for opening this issue again but I noticed that after I got the RampLen() working, that when I change the speed input of the motor, the motor sometimes causes jitter and does not run at the correct speed.

Cheers and have a great weekend
George

MicroBahner, 2020-09-11:
Hi George,
well I don't know what you did to get it working, so I explain what I your sketch is going wrong.
First an explanation for the understanding of ramplen and setSpeed. These two parameters are not independent from each other. Ramplen is the number of steps until the motor reaches the actual set speed, starting from standstill. If xou call setRampLen it references to the actually set speed. Therefor it makes no sense, to set speed to zero, and then set ramplen. This will not work. To make things worse, setting speed to zero is not allowed, it will be changed to the lowest possible speed. And at that low speed a ramp is not possible and makes no sense. Therefore your first setRampLen will be ignored.

Another problem is the fact, that you call the methods setRamplen, setSpeed and rotate very very often. You call them in every loop pass. That's even more often than steps are generated. That makes no sense. You should only call them if the function parameters really change, at least not faster than steps are generated at all.
E.g. you could read your analog values only in fixed time distance, and set the motorparamers only at these times.

gxdeange, 2020-09-11:
Hi Franz-Peter,

Yes, I realised that the SetSpeed and RampLen form the accelleration calculation once I had another look at your examples.

I know my code is very basic but I have only been doing this for a short while and am still learning.

I will try re coding to add some timings (I guess I could use the timeBase for regular analogRead and changes to motor speed parameters.

Cheers and have a great day.
George

closed 2020-09-11 by gxdeange

MicroBahner, 2020-09-12:
H George,
yes, timeBase should be a perfekt solution for this purpose ;) .