MicroBahner/MobaTools #6: Use on Brushed DC Motor driver DRV8870

gxdeange, 2020-10-03:
Hi MicroBahner,

I have been using this library with great success for stepper motors and was curious if you had any intentions to extend the library for use with Brushed DC Motor Drivers and in particular drivers that use 2 PWM signals to control speed and direction (like the DRV8870).

It would be great if this library could control multiple brushed motors in the same non-blocking method for steppers.

Cheers and have a great day
George

MicroBahner, 2020-10-08:
Hi George,
thanks for the suggestion and sorry for the late answer, but I am currently a little handicapped by illness.
I have never dealt with this kind of motor driver before, as I have no use for it. I will have a look at it, but in the short term nothing will happen.  

Best Regards
Franz-Peter

gxdeange, 2020-10-08:
Thanks Franz-Peter,

I hope you get well soon.

I am sure it would be fairly simple as only two values need to be passed to
the motor object (both of them a PWM signal which would provide speed and
direction).

Cheers
George

till234, 2020-11-14:
does this library work for iot boards such as nano33 iot or nodemcu?
thx

MicroBahner, 2020-11-15:
MobaTools works on ESP8266. It does not work on nano33 iot.

MicroBahner, 2021-11-12:
There is no support for brushed DC motors planned.

closed 2021-11-12 by MicroBahner