MicroBahner/MobaTools #8: MoBaTools and older AVR processors?

aikopras, 2021-04-26:
Dear Franz-Peter

First thanks for this great library. 
I not only use it for the default Atmel processors, but also for some older ones (on private boards), such as the ATmega8535, ATmega16 and ATmega32. Unfortunately, it doesn't immediately compile for such processors, due to a different naming scheme for registers, bits and vectors. However, this can easily be fixed, by adding some #defines.

What exactly has to be added, is described at: [https://github.com/MCUdude/MightyCore/blob/master/Library_porting.md]. Until now I have been adding this myself to the MoBaTools.h, and it works. However, after each library update I have to re-add these #defines.

Would it therefore be possible to add these #defines (or just the subset needed by MoBaTools) to the master library?

Thanks

Aiko

MicroBahner, 2021-04-26:
Hi Aiko,
thanks for this hint. I will integrate it to Version 2.4, which will be released soon ( I hope so ;) ). At the moment this is still in development branch.

MicroBahner, 2021-04-26:
Hi again,
you must help me a little bit, because I haven't got these processors.
I installed the MightyCore of your link to get support of theses processors.
If I now e.g. select the ATmega8535
- with version 2.3.1 of MobaTools this processor is never accepted, even with the defines of that link. The reason is clear: the name of the processor is checked and not known.
- with the new version 2.4 ( not yet released, in 'development' branch ) I can successfully compile ( because it only checks some register names, which are obviously defined ) even without the defines of the link. But of course I cannot check if it works.

What did you do?

aikopras, 2021-04-27:
Hi Franz-Peter
Around 10 years ago I started with OpenDCC. First I used the standard OPENDCC hard and software, but soon I started to develop my own PCBs and started to modify the code. After a while the structure of the software remained the same as OPENDCC, but I guess that after a while 80% to 90% was new. Originally I used the ATmega8535, but since RAM was scarce on this processor I moved to the ATMEGA16A. This is still the processor that I use on my existing PCBs, although for novel stuff I use the 328 and 2560.

A couple of years ago I decided to change my original code-base into the direction of Arduino, to improve interoperability between my old (ATMEGA16A) and new (328 and 2560) boards, but also to share code with others (such as the RS-Bus feedback library). And you know how things go, soon you realise that it is better "to stand on the shoulder of giants", instead of doing everything from scratch yourself. So on my servo decoder boards I started to experiment with Mobatools, but immediately run into the problem described above. I therefore made the changes as described on the MightyCore site, after which the servo and button parts of Mobatools worked fine on my ATMEGA16A boards. Attached you'll find an old MobaTools.h file, which worked for me.

If you have a new version of MobaTools and you want me to test that, I would be happy to try the timers, buttons and servo part.  

Bye

Aiko

[MobaTools.h.zip](https://github.com/MicroBahner/MobaTools/files/6382955/MobaTools.h.zip)

MicroBahner, 2021-04-27:
Hi Aiko,
this is a fairly old MobaTools version ;) . But it would be interesting which core you are using to support these processors. The MightyCore that I tried, does not set the \_\_AVR_MEGA\_\_ define. And therefore it will never compile with the versions that check for this define. No matter what registernames are set.

I would apprechiate a lot, if you would test the new version of MobaTools. Independent testers are always welcome.
The beta of the [new version](https://github.com/MicroBahner/MobaTools/archive/refs/heads/Development.zip) can be found on github in the [development branch](https://github.com/MicroBahner/MobaTools/tree/Development). This version is actually finished, and it's just a matter of finding any bugs.

aikopras, 2021-04-27:
Hallo Franz-Peter

Yes, that is indeed an old version ;-). But one of the lessons I learned is "never break something if it doesn't need to be fixed" ;-)

I just tested it, and V2.4 development now compiles and runs "out-of the box". See attached info file. Note that I only checked the servo behaviour, but could test in the next few days the timers as well. I don't see how I can easily test the steppers, however.

The correct __AVR_MEGA__ define is set by the MightyCore package, but you have to select in the Arduino IDE under Tools the MightyCore ATMega16 board. At least this worked for me. I know I played with all kind of settings in the past, but I don't believe it only works here since there are some "legacy settings left over from the past". But if you want, I can check deeper.

Anyway, thanks already for the quick reaction and I would be happy to continue testing with the new development version.

Herzliche Grüsse, Aiko

[test-servo-MoBaTools.ino.zip](https://github.com/MicroBahner/MobaTools/files/6386764/test-servo-MoBaTools.ino.zip)

MicroBahner, 2021-04-28:
Hi Aiko,
the difference between V2.4 and the previous versions is the way how it is checked if the actual processor is supported. The use of registernames ( with avr-processors ) is the same. Obviously th registernames are the same in standard core and MightyCore.
If in V2.4 an AVR processor is detected (  **ARDUINO_ARCH_AVR** is defined )  it only checks some registernames to be shure there is a 16-bit timer 1. 
The previous versions checked only if **\_\_AVR_MEGA\_\_** is defined ( this define is set by the standard arduino core, but not by the MightyCore ).
It would be fine if you do some more testing with the new version. Of course you need only test what is easy for you to test.

Vriendelijke groeten  ;)
Franz-Peter

MicroBahner, 2021-11-12:
Closed because no further problems have been reported.

closed 2021-11-12 by MicroBahner