MicroBahner/MobaTools #9: Support planned for newer processors / boards?? (4809, Nano Every, AVR Dx)

aikopras, 2021-07-14:
Dear Franz-Peter

Thanks for adding support for some older ATMEGA processors. But now I have the opposite question. Do you have any plans to also support newer processors, such as the 4809 (which is included on the Nano Every) or even the AVR Dx processors? 

The boards for these processors have been made available by MCUdude and SpenceKonde:
- https://github.com/MCUdude/MegaCoreX
- https://github.com/SpenceKonde/DxCore

Certainly the 4809 seems to be quite interesting vor people developing their own "Arduino compatible" hardware boards. The 4809 is not only cheaper than the 328, but also more powerful. 

Thanks

Aiko

SkipperDo, 2021-07-15:
Hi there...

Is it planned to port this Libary to the Raspberry Pi? - This would be very cool!!

MicroBahner, 2021-07-15:
Hi, Aiko,
yes, support for more processors is planned, but it takes a little time because arduino is not my only hobby ;)
MobaTools have a big problem with 4809, because it does not really allow nested interrupts. But a few days ago someone presented a solution for this problem in the arduino forum. Maybe this is a solution for MobaTools too. I'll have a closer look at this the next time.
Greetings
Franz-Peter

MicroBahner, 2021-07-15:
Hi SkipperDo,
as far as I know, Raspberry PI is based on a full featured linux. This is a completely different development environment with completely different API. I think a port is not possible, that would be a complete new development.

aikopras, 2021-07-15:
Hi Franz-Peter
I'm now playing myself with the 4809 and similar processors. There are several differences which took me some time to understand, but I find that the peripheral handling is now solved in a more elegant way. So that is basically good news.

Just for curiosity, what is the issue regarding nested interrupts? I assume that many ISRs are relatively short. I know, however, that DCC decoding routines, such as the NMRA library, may take quite some time and I could understand that servo timing, as needed in your code, should be very precise to avoid jitter. I'm not familiar enough with your code, however, to know if that is true. 

Keep enjoying all hobbies :-)
Aiko

MicroBahner, 2021-07-17:
Hi Aiko,
Nested interrupts are needed especially for the stepper ISR on relatively slow processors, like the 8-bit Arduinos. If there are severals steppers running at the same time, and also softleds active, this ISR may take fairly long ( up to 200µs ). This is too much for some other ISR, like servo or the NmraDcc lib. These Interrupts must be able to interrupt the stepper ISR.
Unfortunately the 4809 allows to set only one ISR to a HIGHER priority. But it is not possible to set an ISR to a LOWER priority.
Greetings
Franz-Peter

aikopras, 2022-01-04:
Hi Franz-Peter

Any progress? ;-)
In the last months I've spend quite some time "playing" with the MegaCoreX and DxCore processors. My RS-Bus and DCC decoding libraries are now not only running on the new 4809 and AVR-DA128 controllers, but (in my opinion) are (thanks to the improved timers and the event library) also running much better on these newer (and cheaper) controllers. 

In the past I looked at your servo library, but to be honest I don't see through if and where changes should be made. From my experience with my own libraries I also learned that a one-to-one copy to the new timers of MegaCoreX and DxCore processors may not give satisfactory results. See for example: [https://github.com/aikopras/AP_DCC_library/blob/main/extras/Performance_MegacoreX.md](https://github.com/aikopras/AP_DCC_library/blob/main/extras/Performance_MegacoreX.md)

Let me therefore ask a different question. What are the main differences between your MobaTools servo library, and the standard servo libraries that come with Arduino? For MegaCoreX and DxCore these servo libraries have already been updated to the new processors. I assume the main differences are:
- Your library supports multiple servo's
- Your library supports that slow acceleration and de-acceleration.

Do I miss something?

Thanks
Aiko :-)

MicroBahner, 2022-01-06:
Hi Aiko,
yes the main difference ist the possibility to set the speed of the servo. 
And it can handle up to 16 servos with one timer - but it can only use one timer, so 16 servos is the max.

Im working on support for the 4809, but at the moment things move on slowly ... :-( 
There are simply too many other projects ;-).

Regards
Franz-Peter

MicroBahner, 2023-08-14:
Hi Aiko,
currently I'm working on Version 2.5 of MobaTools, and there is a Beta release in branch [Development2.5.0](https://github.com/MicroBahner/MobaTools/tree/Development2.5.0). The Mega4809 is supported in this release.
Regards, Franz-Peter

aikopras, 2023-08-17:
Hi Franz-Peter
Nice :-)
I quickly scanned through the 2.5 branch, and got the impression that you rely on the "default Arduino" 4809 board definition, and not on MCUdude's MegaCoreX board. Is that a deliberate choice? The MegaCoreX board offers many enhancements, such as support for Events. Code that runs on the MegaCoreX board is generally upwards compatible with the DxCore boards, which support the new AVR DA and DB processors. Some of them have multiple TCA timers (I saw that you use TCA0). For my DCC and RS-Bus code, I rely on some of the TCB timers to make DCC signal detection and RS-Bus feedback more robust, with a lower code footprint. 
So in short, by moving to MegaCoreX and DxCore you get the richness of all new AVR processors :-)
Bye, Aiko

MicroBahner, 2023-08-17:
Hi Aiko,
I deliberately wrote it to run with the standard Arduino core. However, it also works with the MegaCoreX. But I didn't want it to run **only** with the MegaCoreX. Also he MegaCoreX does not (yet) work with the UNO WiFi R2. There are errors when uploading the sketch.
Greetings, Franz-Peter

MicroBahner, 2023-09-08:
Hi Aiko,
yes these processors look interesting. But my time is limited, and first I have to release the V2.5. 
And the next planned step is to support the Arduino UNO R4, which seems to be the next widely used Arduino.
Do you know of Arduino compatible boards with a DxCore processor?
Franz-Peter

aikopras, 2023-09-09:
Hi Frans-Peter
I understand that time is limited and you have to make choices. I also have many interesting ideas of what I would like to do, and at the end am able to implement only 10% ;-)

the choice to first go for the UNO R4 is understandable and might be the better choice. Most people want to use a ready to buy board, and don’t want to build boards themselves. 

I don’t know any Arduino DxCore boards, but of course Microchip sells similar boards for experiments (the AVR curiosity boards). At this moment I have built some servo boards for DxCore; these boards still need a separate DIL AVR 28 since at the time I ordered them at JLCPCB I couldn’t get AVRs from them. Nowadays that is no longer a problem, and I can order boards with AVRs already assembled. I’m planning to build some in the near future, to control some Microsteppers for certain “Formsignale” ;-). I guess you know (some of) these signals even better then I do ;-) ;-)

Anyway, if you want one of my current or new DxCore board for experimenting, please let me know and I will happily send you one (for free, off course) :-)
Bye, Aiko

MicroBahner, 2023-09-12:
Hi Aiko,
thanks for your offer, but I think in the near future I will dive deeper into the R4. And my new model railroad also needs some more time. It is a little narrow gauge layout with only one small station - there are no signals anymore ;-) . To operate them with microsteppers is interesting - I always used servos.

I now released version 2.5.

Greetings, Franz-Peter

MicroBahner, 2024-09-05:
New arduino boards will be added from time to time .... ;-)

closed 2024-09-05 by MicroBahner