MicroBahner/MobaTools #10: Compiler warning

aikopras, 2022-01-04:
Hallo Franz-Peter

While compiling with all warnings on, I get the following compiler warning:
```
... MoToServo.cpp:314:14: note: in expansion of macro 'constrain'
... warning: comparison between signed and unsigned integer expressions
```

As far as I know, the cause is that the compiler selects an unsigned type for MINPULSEWIDTH and MAXPULSEWIDTH, whereas the type of _minPw and _maxPw is uint16_t.
The warning can easily be avoided by adding the uint16_t to the #defines in MobaTools.h. 
For example (for the AVR and STM32 case):
```
#define MINPULSEWIDTH   700     // don't make it shorter than 700
#define MAXPULSEWIDTH   2300    // don't make it longer than 2300
```
should be changed into:
```
#define MINPULSEWIDTH   ((uint16_t)700)     // don't make it shorter than 700
#define MAXPULSEWIDTH   ((uint16_t)2300)    // don't make it longer than 2300
```
I know this is a minor thing, but I always try to avoid red lines in the compiler output ;-)

MicroBahner, 2022-01-06:
Hi Aiko,
how did you manage to get this warning? I always compile with all warnings on, but I did not get this warning.

Franz-Peter

MicroBahner, 2022-04-28:
Fixed in release 2.4.3

closed 2022-04-28 by MicroBahner