MicroBahner/MobaTools #11: 2 noob questions

trombose009, 2022-01-14:
Hi, 

1.) if the stepper is moving at higher speeds, will it stop instantly if it has to stop? Or does it go on a few steps  because it is not able to stop imediately comparing to lower speeds?

2.) I am using the A4988 and an esp8266 to control one stepper motor. I am not sure how to use the attachEnable thing to disable the stepper until the next moving action. 
Is there a wiring and/or a sketch example where it is shown how to do that?

greetings

MicroBahner, 2022-01-15:
1.) Depends on the ramp configured and on your mechanics. And how you stop it. The 'stop' method is an emergency stop and instantenously stops creating steps. Wether your stepper can follow this and really stop immediately depends on the speed an your mechanics. At higher speeds most likely not. If you define a ramp and let the lib stop the stepper at the target position it will decelerate until stop and not loosing steps.

2.) connect a GPIO-Pin to the EN Pin of the A4988 and put the command
`myStepper.attachEnable( GPIOPin, 50, LOW); //50 is the time (in ms ) between end of stepper movement and disabling it.`
into setup(). You must not use GPIO16 as the dir pin in this case.

trombose009, 2022-01-15:
Great, thank you very much, it works. 
Another question.
If I start my device, it moves to an endstop switch. in setup() I declare
myStepper.setSpeed( 300 ); 
what can I do to make it moving at lower speed only in the nullposition() function?
It seems not to work if I just make this:
`void nullposition() {

  //er soll 2600 in eine Richtung fahren bis zum Endschalter. So wird die Nullposition ermittelt.
  Serial.println("Nullposition einnehmen");
myStepper.setSpeed( 30); 
   steps = "-2600"; //-2600
  myStepper.doSteps(steps.toInt());
  positionvalue = 0;
myStepper.setSpeed( 300 ); 
 }`

MicroBahner, 2022-01-16:
You can set the speed at any time, so this should work. But
`myStepper.setSpeed( 30);` is fairly slow, it means only 3 rev/min ( if you set the steps/rev correctly ). If you want to set the speed in terms of steps you must use `setSpeedSteps(stepsPer10sec)`

And I cannot see that you check your endswitch to determine the zero position, and you did not set the zeroposition for 'myStepper'.
This:
[code]
steps = "-2600"; //-2600
myStepper.doSteps(steps.toInt());
[/code]
is a little bit strange too. Why not directly set the steps to -2600?

trombose009, 2022-01-16:
My slider should just move to it's starting position everytime the device is switched on. The endstop switch is a physical switch which is touched by the slider. This part ist scripted in another function. I just want to know how I can set the moving speed low during this initialization process so that the slider does not crashes into my switch at high speed. The number of steps to do this is not known cause the slider may have moved manually during the device is switched off.

trombose009, 2022-01-16:
THanks a lot, it is working now! Great!

closed 2022-01-16 by trombose009