MicroBahner/MobaTools #13: 1/4 StepMode 800 full rotation

creativesamurai1982, 2022-04-30:
Hi there. A user in the arduino forums (StefanL38) pointed me to your library. Amazing by the way.
But im having trouble setting up 1/4 stepping. Is it available with MoBaTools? or am I 
going to have to set me drive to HALF Step. which i really dont want to do.

Dave

MicroBahner, 2022-04-30:
Hi Dave,
just like Accelstepper the MoToStepper class doesn't care about microstepping. If you comtrol a stepper with a step/dir driver you always use STEPDIR when creating the object. The steps/rev is needed if you want to set the speed in rev/min or want to define a target position in degree. Of course this depends on the microstepping set at the driver, the base steps per rev of the motor and maybe a gear.
FULLSTEP or HALFSTEP when creating the object is only used with unipolar steppers that are controlled by 4 pins.

Franz-Peter

creativesamurai1982, 2022-04-30:
Nice one. Thanks for the info. I'm finding your lib easier to use and it seems more accurate when stopping at a 45 degree angle 1.8 that way when I turn the stepper on and off via enable there no wobble with the motor trying to find a full step to land on. Great work.

closed 2022-09-28 by creativesamurai1982