MicroBahner/MobaTools #15: Question regarding MoToTimer

aikopras, 2022-07-17:
Hi Franz-Peter
I was looking at your MoToTimer class but somehow fail to understand why you need two flags (RUNNING s well as NOTEXPIRED), instead of just a single one (RUNNING). Is there something that I fail to understand?

Also I found the line
`    if ( active & RUNNING ) active &= ~RUNNING | ( millis() - startTime < runTime );` 
a bit unclear.

 If I understand it well, the part `active &= ~RUNNING` makes sure that the RUNNING bit gets cleared, whereas the NOTEXPIRED bit remains untouched. Subsequently you set the RUNNING bit again, depending on the value of `( millis() - startTime < runTime )`. 

Wouldn't it be a clearer way be to write this as:
`     status = ( status & NOTEXPIRED ) | ( millis() - startTime < runTime );`

 Bye, Aiko

MicroBahner, 2022-07-17:
Hi Aiko,
the flag 'NOTEXPIRED' es needed for the method 'expired()'. This method returns TRUE only once after the timer expired. In contrast to 'running', which returns TRUE **while** the timer is running, the expired method returns TRUE **only once**, after the time has elapsed. Therefore after the time has elapsed ( runing() would return FALSE ) you must remember if 'expired' has already been called.

Regards, Franz-Peter

aikopras, 2022-07-17:
Hi Franz-Peter
I understand that you want the expired() method to return TRUE only once. But do you need two flags for that?
The following code has only one flag, and should, as far as I know, work equally well (and has an even smaller code footprint). Or do I miss something??

```
class MoToTimer {
  private:
    bool active = false;
    unsigned long startTime = 0;
    unsigned long runTime = 0;

  public:
    void setTime(unsigned long value) { runTime = value;}

    void start() { 
      startTime = millis();
      active = true; 
    }   

    void stop() {active = false;}
    
    bool expired() {
      // Only the first call after expiration returns true.
      // If stop() was called before expiration, false will be returned
      if (active) {
        if (millis() - startTime > runTime) {
          active = false;
          return true;
        }
      }
      return false;
    }

    unsigned long getRuntime() {return runTime;}

    unsigned long getElapsed() {
      if (active) return ((millis() - startTime) + 1);
        else return runTime;
    }
    
    unsigned long getRemain() {
      if (active) return runTime - (millis() - startTime);
        else return 0;
    }
};

```
Note that I changed the semantics of setTime(), that in my version doesn't automatically start the timer.

Bye, Aiko

MicroBahner, 2022-07-17:
Hi Aiko, I think it is a little bit more complicated than your version. Your version misses the running() method. And you need this function - not only as external method, but also internally to check wether the timer is still running when you call getElapsed() or getRemain(). You cannot check that with the active flag only, if the flag is not reset when the time is over.

Regards, Franz-Peter

MicroBahner, 2022-07-18:
Hi Aiko,
the basic problem remains the same in my opinion. In the new code you force the user to use '`expired()`', because only there the flag is reset. Depending on the application, it may be that only '`running()`' is used, or only `getElapsed()` or `getRemain(). `The method '`expired()`' - with the 'state change detection' is only one of the methods to query the timer. It must also work if only one of the others is used. Therefore, in my code, `running()` is called internally for all methods (`expired, getRemain, getElapsed`) in order to always reliably recognise the expiry of the timer and to reset the 'active' flag. The 2nd flag is then only necessary for expired() to ensure the 'state change detection'.
Regards, Franz-Peter

aikopras, 2022-07-18:
Hi Franz-Peter
I'm getting confused now :-)
For me that generally means that I want to understand it :-)

In your code the `RUNNING` flag (bit 0) is equivalent to `millis()-startTime < runTime`, unless `stop()` is called before. 
If `stop()` is called before, the `RUNNING` flag will be cleared, as well as the `NOTEXPIRED` flag.
The `RUNNING` flag is therefore superfluous, and can be replaced by checking `millis()-startTime < runTime` in combination with the `NOTEXPIRED` flag.

Indeed my code only clears the `NOTEXPIRED` flag after `expired()` or `stop()`. But what is the problem with that? Why would I need to 
> always reliably recognise the expiry of the timer?

Since I got rid of the `RUNNING` flag, there is no need anymore to 

> reset the 'active' flag.

I still believe that your methods and my methods deliver the same outcome in all cases.
However, I'm old and experience enough to know that I may be caught in a "tunnel vision", and therefore I'm checking with you. I hope you don't mind.

Maybe I should therefore ask the question differently. Do you believe that there are circumstances in which your code and mine give different results? If yes, can you elaborate.
Bye, Aiko

MicroBahner, 2022-07-18:
Hi Aiko,
as an example: If you call your function 'getRemain'
```
    unsigned long getRemain() {
      if (notExpired) return runTime - (millis() - startTime);
        else return 0;
    }

```
after the time has elapsed and without calling expired() in advance, you will get a huge number as return value. Because the `notExpired` flag has not been reset, the return value ist
` runTime - (millis() - startTime);`
After the timer has expired this will evaluate to a negative number. Because the type of the function is `unsigned long` this will be treated as a very big number.

It's true, that the 'active' flag could always be replaced by checking ` millis()-startTime < runTime`. But that would produce more code. It may be possible to work without the 'active' flag, but not by simply dropping all checks of the active flag.

MicroBahner, 2022-07-18:
Hi,
another problem, why the active flag is indeed NOT the same as `millis()-startTime < runTime`:
if you stop the timer prematurely with stop(), the active flag will be reset, but `millis()-startTime < runTime` would still return true. stop() couldn't be as easy without the active flag.

aikopras, 2022-07-18:
Hi Franz-Peter
Ah, yes, you are right. getRemain() and getElapsed needs to also check if the time has not yet elapsed. So it is indeed good to include running() as check. This gives the following:

```
    unsigned long getElapsed() {
      if (running()) return ((millis() - startTime) + 1);
        else return runTime;
    }

    unsigned long getRemain() {
      if (running()) return runTime - (millis() - startTime);
        else return 0;
    } 
```

>... the 'active' flag could always be replaced by checking millis()-startTime < runTime. But that would produce more code.

My intuition is the same as yours, that my approach would result in more code.
However, I made some very simple tests (see the attached test code below):
1) AVR128DA48: 
    - MoToTimer: 974 bytes Program  / 13 Bytes RAM
    - New timer: 850 bytes Program  / 13 Bytes RAM
2) Arduino UNO:
    - MoToTimer: 1132bytes Program  / 18 Bytes RAM
    - New timer: 1014 bytes Program  / 18 Bytes RAM

I've also looked at the generated assembly, and observed that the compiler sometimes makes optimisations that I hadn't expected. These optimisations are context depended, and may vary from case to case. Still I can't really explain the difference between both versions. One element is that my new code didn't have the class constructor, which saves roughly 60 bytes. But that is only 50% of the difference. But anyway, it seems that the new timer code is certainly not longer, and likely even shorter.

Bye, Aiko
[testcode.zip](https://github.com/MicroBahner/MobaTools/files/9133751/testcode.zip)

aikopras, 2022-07-18:
Bye
> another problem, why the active flag is indeed NOT the same as millis()-startTime < runTime:
if you stop the timer prematurely with stop(), the active flag will be reset, but millis()-startTime < runTime would still return true. stop() couldn't be as easy without the active flag.

In my code `stop() `includes: `notExpired = false;`
That should have the same result.

closed 2022-07-26 by aikopras