MicroBahner/MobaTools #16: Direction control

creativesamurai1982, 2022-09-28:
Hi there, I'm posting here as I don't know how else to contact you.
I've been playing around with your library for some time now, and I must tell you how impressed and how useful you're library is. 
But on to my question.
To control my steppers I'm sending commands via serial
F200 is forwards 200 steps. But I'm having issues working out B200 back 200 steps. To go back I know to use the - minus sign (stepper1.doSteps(-noOfSteps1*16/5); 
If (dir == "B") {//go back}
If (dir == "F") {//go forward}
You'll find my code in my github if it helps. 
Again, I'm sorry if I shouldn't post here, kinda new to github and programming in general. 

Dave.

MicroBahner, 2022-10-02:
HI Dave,
I'm on vacation at the moment and can't really check that. I'll answer again, if I'm back.
Franz-Peter

drf5n, 2023-04-17:
Is there a method to determine the direction that a stepper is turning?  Sort of like the functionality of:

     long signedSpeed = ((long)myStepper.getSpeedSteps())*(digitalRead(dirPin)?1:-1);
     if (signedSpeed > 0 ) Serial.println("Moving clockwise");
     if (signedSpeed < 0 ) Serial.println("Moving counterclockwise")

myStepper.getSpeedSteps() seems to be non-negative, whether or not it is moving forward or backward.

MicroBahner, 2023-04-23:
Unfortunately there is no method to determine the direction of turning at the moment. But it may indeed be useful. I'll put it on my todo list ;)

MicroBahner, 2023-06-21:
Hi,
currently im working on Version 2.5 of MobaTools. There is a branch ( [Development2.5](https://github.com/MicroBahner/MobaTools/tree/Development2.5.0) ) which represents the actual development state. Now the value of getSpeedSteps() also indicates the direction by it's sign. Negative values indicate going backwards.

MicroBahner, 2023-09-12:
Solved with release 2.5

closed 2023-09-12 by MicroBahner

Creative-Samurai1982, 2023-09-12:
Nice work, thank you.