MicroBahner/MobaTools #17: Wish, it could run on Teensy 3.5 or 4.1

Piki55, 2022-12-20:
Hello Franz-Peter,
i wish this absolut "Geile" Lib would run on Teensys.
I am building a Boeing B738 Homecockpit, and use >10 Teensys 3.5 and 4.1 in the Setup.
As far as i can read your Manual, this could be the absolute best Lib for my Purposes.
May be, you have the Time to implement those Teensys.

merry Christmas

Gerhard

MicroBahner, 2022-12-20:
Hello Gerhard,
unfortunately I don't have the time and possibility to further develop the MobaTools library at the moment. I think it will take a few months until I can develop on MobaTools again.
But if I'm not mistaken, there is a very sophisticated stepper library for Teensies.

Many greetings and a Merry Christmas
Franz-Peter

Piki55, 2022-12-21:
Hallo Franz-Peter,
schade, your Combination of Tool is the interresting Thing, not a Stepperlib only.
But, Thank you and Merry Christmas from Hannover

closed 2023-04-23 by MicroBahner