MicroBahner/MobaTools #18: Error message with Heltec esp32 Wifi Kit

blue0ceanby, 2023-01-28:
I use the Mobatools on different Arduino and ESP32 platforms. Now I wanted to make a project wth an ESP32 Wifi Board.

I use the Mobatools on different Arduino and ESP32 platforms. Now I wanted to make a project with an ESP32 Wifi Board. I copied the whole error message list I hope you can work with it. If necessary I can also send you a board for testing.

```
In file included from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/esp32/drivers.h:7,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/utilities/MoToBase.h:23,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/MobaTools.h:169,
                 from C:\Users\xxxx\AppData\Local\Temp\.arduinoIDE-unsaved2023028-2740-16plkyv.94fu\Treppenlicht\Treppenlicht.ino:21:
c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/esp32/esp32-mt-spi.h:28: warning: "FSPI" redefined
 #define FSPI  1 //SPI bus attached to the flash (can use the same data lines but different SS)
 
In file included from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal.h:83,
                 from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/Arduino.h:36,
                 from C:\Users\xxxx\AppData\Local\Temp\arduino-sketch-24AC3701DBFD0EE6AB5E5F4CDBA15E94\sketch\Treppenlicht.ino.cpp:1:
C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-spi.h:29: note: this is the location of the previous definition
 #define FSPI  0
 
In file included from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/esp32/drivers.h:7,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/utilities/MoToBase.h:23,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/MobaTools.h:169,
                 from C:\Users\xxxx\AppData\Local\Temp\.arduinoIDE-unsaved2023028-2740-16plkyv.94fu\Treppenlicht\Treppenlicht.ino:21:
c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/esp32/esp32-mt-spi.h:29: warning: "HSPI" redefined
 #define HSPI  2 //SPI bus normally mapped to pins 12 - 15, but can be matrixed to any pins
 
In file included from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal.h:83,
                 from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/Arduino.h:36,
                 from C:\Users\xxxx\AppData\Local\Temp\arduino-sketch-24AC3701DBFD0EE6AB5E5F4CDBA15E94\sketch\Treppenlicht.ino.cpp:1:
C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-spi.h:30: note: this is the location of the previous definition
 #define HSPI  1
 
In file included from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\drivers.h:7,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:5:
c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\esp32-mt-spi.h:28: warning: "FSPI" redefined
 #define FSPI  1 //SPI bus attached to the flash (can use the same data lines but different SS)
 
In file included from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal.h:83,
                 from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/Arduino.h:36,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:4:
C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-spi.h:29: note: this is the location of the previous definition
 #define FSPI  0
 
In file included from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\drivers.h:7,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:5:
c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\esp32-mt-spi.h:29: warning: "HSPI" redefined
 #define HSPI  2 //SPI bus normally mapped to pins 12 - 15, but can be matrixed to any pins
 
In file included from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal.h:83,
                 from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/Arduino.h:36,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:4:
C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-spi.h:30: note: this is the location of the previous definition
 #define HSPI  1
 
In file included from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal.h:83,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\esp32-mt-spi.c:17:
C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-spi.h:29: warning: "FSPI" redefined
 #define FSPI  0
 
In file included from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\esp32-mt-spi.c:16:
c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\esp32-mt-spi.h:28: note: this is the location of the previous definition
 #define FSPI  1 //SPI bus attached to the flash (can use the same data lines but different SS)
 
In file included from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal.h:83,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\esp32-mt-spi.c:17:
C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-spi.h:30: warning: "HSPI" redefined
 #define HSPI  1
 
In file included from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\esp32-mt-spi.c:16:
c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\esp32\esp32-mt-spi.h:29: note: this is the location of the previous definition
 #define HSPI  2 //SPI bus normally mapped to pins 12 - 15, but can be matrixed to any pins
[... 441 lines cut ...]
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/MobaTools.h:169,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\utilities\MoToStepper.cpp:13:
C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-spi.h:29: note: this is the location of the previous definition
 #define FSPI  0
 
In file included from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/esp32/drivers.h:7,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/utilities/MoToBase.h:23,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/MobaTools.h:169,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\utilities\MoToStepper.cpp:13:
c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/esp32/esp32-mt-spi.h:29: warning: "HSPI" redefined
 #define HSPI  2 //SPI bus normally mapped to pins 12 - 15, but can be matrixed to any pins
 
In file included from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal.h:83,
                 from C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/Arduino.h:36,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/utilities/MoToBase.h:11,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src/MobaTools.h:169,
                 from c:\Users\xxxx\OneDrive\Dokumente\Arduino\libraries\MobaTools\src\utilities\MoToStepper.cpp:13:
C:\Users\xxxx\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-spi.h:30: note: this is the location of the previous definition
 #define HSPI  1
```
 
'
exit status 1

Compilation error: exit status 1

MicroBahner, 2023-03-16:
Sorry for the very late reply. But unfortunately I have very little time at the moment (actually none at all :-(  ) to take care of the further development of MobaTools. I hope that I will have more time to deal with error messages from around May onwards.

MicroBahner, 2023-06-22:
Hi,
currently im working on Version 2.5 of MobaTools and I suppose your problem is the same as issue #5. There is a branch ( [Development2.5](https://github.com/MicroBahner/MobaTools/tree/Development2.5.0) ) which represents the actual development state. I was able to compile an example sketch for the board 'Heltec WiFi Kit 32' without error. So your issue may be solved. It would be nice if you could test your sketch with this development version.

MicroBahner, 2023-09-12:
Solved with release 2.5