MicroBahner/MobaTools #19: Error compiling for board ESP32 Dev Module

Roboman142, 2023-03-23:
Using Mobatools library works fine with an Arduino board board when I try to upload my code to and ESP32 I get "Error compiling for board ESP32 Dev Module".

[esp arduino error.txt](https://github.com/MicroBahner/MobaTools/files/11054837/esp.arduino.error.txt)

MicroBahner, 2023-04-23:
This is a problem with newer ( V2.x ) versions of ESP32 boards manager. With V1.06 of ESP32 boards manager it works. I think there will be a solution in the next main version of MobaTools. But it will surely take a while...

MicroBahner, 2023-06-21:
Hi Stefenzo,
currently im working on Version 2.5 of MobaTools. There is a branch ( [Development2.5](https://github.com/MicroBahner/MobaTools/tree/Development2.5.0) ) which represents the actual development state. Your issue should be solved already. But intensive testing is still necessary.

MicroBahner, 2023-09-12:
Solved with release 2.5

closed 2023-09-12 by MicroBahner