MicroBahner/MobaTools #20: Problems with zero speed

Stefenzo2, 2023-04-07:
Really great library,

I have a little problem with the stepper motor control. I calculate my speed based on the value of an analog input. Thereby it can come with very low values with the calculation to zero. If I now set the speed to 0, something hangs and nothing works for a few seconds. I have now placed a small query before setting the speed, which replaces zero with one. The problem is solved. Perhaps one could catch this however also directly in the library.

I use an Arduino Mega 2560.

Thank you for providing this great tool.

Greetings Stefenzo

MicroBahner, 2023-04-23:
Hi Stefenzo,
yes, there is a problem with setting speed to zero, setSpeed is not meant to stop the motor. MoToStepper is designed as a positioning tool. And that means the motor only stops, if it has reached its target position. Of course you can set the target position to the actual position at any time to stop the motor.
I know about this problem with setting speed to zero and probably there will be a solution in the next main version.
Greetings

MicroBahner, 2023-06-21:
Hi Stefenzo,
currently im working on Version 2.5 of MobaTools. There is a branch ( [Development2.5](https://github.com/MicroBahner/MobaTools/tree/Development2.5.0) ) which represents the actual development state. Your issue should be solved already for all boards exept ESP8266. But intensive testing is  still necessary.
Setting speed to 0 leads to ramping down and standstill, irrespective wether tha target position has been reached or not. Setting speed higher again will move on to the last set target. You can also change the target positon during standstill.

closed 2023-09-12 by MicroBahner

Stefenzo2, 2023-09-12:
> Solved with release 2.5

Thank you👍