MicroBahner/MobaTools #22: AltSoftserial

dunno1093, 2023-10-26:
My Project does not compiler when using mobatools and AltSoftserial together. Both seem to have a variable called vector _12. Is there a possibility to solve this Problem?

MicroBahner, 2023-10-27:
Hi dunno1903. I suppose you tried this on an UNO R3? The problem is a hardware confict as both libraries need a 16-bit timer and on an UNO R3 there is only one. You cannot use both of these two libraries in parallel on this board.

dunno1093, 2023-10-27:
Thank you for the quick reply. Yes,  UNo R3. Then i Look for a Board with more timers or try to use softserial library instead.

MicroBahner, 2023-10-27:
The Leonardo may be an alternative. MobaTools uses timer3 on this board ( hopefully AltSoftSerial sticks to timer1 there ;-) ).
On the other hand - the HW serial line on pins 0/1 can be used independently of the USB connection on this board. So you may even not need AltSoftSerial there ... ?

dunno1093, 2023-10-27:
OK, Sounds good..i'll give it a try...thanks a lot

MicroBahner, 2023-10-27:
You're wellcome

closed 2024-02-17 by MicroBahner