MicroBahner/MobaTools #23: Using with shift in/out libaries like shifty

mavoIn, 2023-12-18:
Hello,

First of all I would like to thank you for your very useful lib here.
I hope it's okay if I ask here, as there don't seem to be any new registrations confirmed in the Stimmies Modellbahnforum at the moment.

Is there a simple way to control or read out the SoftLEDs or SoftButtons via a shift register.

Currently I would like to use a lib like https://github.com/johnnyb/Shifty. However, I'm not that experienced and wanted to ask if anyone has already done it and could maybe share his code with me?

Thank you very much.

MicroBahner, 2023-12-18:
Hi,
I'm not quite shure what you mean with 'softButtons?.
SoftLeds mus be connected directly to a pin on the Arduino.
The MoToButtons library allows to connect the buttons in an arbitrary way. You can connect them via shift registers, I2C, matrix - as you like. This is possible because the raw reading of the buttons can be done in a callback function. And it is up to you to write this function.
I never used the lib you linked.
best regards and a merry christmas :-)

mavoIn, 2023-12-18:
Thx for the quick answer. This will help me... unfortunately i have to switch the library to be able to trigger leds via shift register. :-(
Merry christmas ;-)

closed 2023-12-18 by mavoIn