MicroBahner/MobaTools #24: "function __vector_11" Problems

saturnquack, 2024-01-29:
Hello! I am trying to make a code simulation in Wokwi, and when I attempt to run my code it gives me a build error with the following text:
`MoToServo.cpp.o (symbol from plugin): In function `__vector_11':
(.text+0x0): multiple definition of `__vector_11'
/libraries/Servo/avr/Servo.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status

Error during build: exit status 1`

I have absolutely no clue what this means, or how to solve it. Is there a fix or some way to solve it? I'll attach all my code below here and a screenshot of what the error looks like. Thank you in advance for any help. 

[image]
`
#include <Servo.h>
#include <MobaTools.h>

const int trigPin = 9;
const int echoPin = 10;
long duration;
int distance;
MoToServo myservo;
void setup() {
  pinMode(trigPin, OUTPUT); 
  pinMode(echoPin, INPUT); 
  Serial.begin(9600); 
  myservo.attach(3); 
  myservo.setSpeed(100); 
  
}

void ultrasonic() {
  digitalWrite(trigPin, LOW);
  delayMicroseconds(2);
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
  duration = pulseIn(echoPin, HIGH);
  distance = duration * 0.034 / 2;
  Serial.print("Distance: ");
  Serial.println(distance);
}

void loop() {
    ultrasonic();
    if(distance <= 200)
    myservo.write(random(0, 180));
    delay(300);

    
}`

https://wokwi.com/projects/387831943535863809

MicroBahner, 2024-01-29:
Hi,
You cannot use the Servo.h lib together with MobaTools.h because both use timer1 interrupts.
If you use the MoToServo class you need not ( and must not ) include Servo.h

closed 2024-01-29 by saturnquack

MicroBahner, 2024-01-29:
You're welcome