MicroBahner/MobaTools #25: attachEnable does not DISABLE the stepper after move

pandel, 2024-02-15:
Hi!

I am using a Arduino UNO Rev3 a TB6600 in **common anode** configuration and setting everything up via

```
const byte stepPin = 11;
const byte dirPin  = 12;
const byte enaPin  = 13;
...
myStepper.attach( stepPin, dirPin );
myStepper.attachEnable( enaPin, 10, HIGH );        //  ( HIGH=active )
```

The stepper is working fine so far, but it does not disable at the end. I tried different "manual" methods, like

- `digitalWrite(enaPin, 0)`
- send an explicit `myStepper.stop()` after having reached the position
- `myStepper.moveTo(1)` (after reaching and setting a zeroPoint)

but nothing works. I am using the current version available via PlatformIO download: 2.6.1

What can I do?

Regards,
Holger

pandel, 2024-02-15:
Sorry for bothering you... I don't know what happened, but I cleaned up the build directory and tried again with `myStepper.stop()` and now it works. I'd love to know why now and not before, but I am happy that it works in the end.

closed 2024-02-15 by pandel

MicroBahner, 2024-02-15:
Hi Holger,
how did you wire the enable input of the TB6600? Pin 13 is also the internal LED of the UNO. Does it switch when the stepper is started/stopped?

> BTW, are there any (undocumented) possibilies to enable/disable a stepper directly? 

Of course you can directly set/reset the enable pin - independent from the lib. But if you do so while the stepper is running, it will stop immediately without the library knowing about that. But you can try this to see if your wiring is ok.
Regards,
Franz-Peter

MicroBahner, 2024-02-15:
> and tried again with myStepper.stop() and now it works.

You should not need an explicit `myStepper.stop()` to disable the stepper. It should always be disabled when it is not running. No matter if by a stop comand, reaching the target or rotate(0) ...

pandel, 2024-02-15:
@MicroBahner Hey Franz-Peter! Ich sag's mal auf Deutsch: ich bin ein Vollidiot! Ich hatte mit Example Code rumgespielt und da war tatsächlich noch eine separate Ansteuerung der LED an Pin 13 mit drin. Da ich diese LED nie benutze und somit die Verbindung zu Pin 13 dafür überhaupt nicht auf dem Schirm hatte, hab ich daran gar nicht gedacht - tatsächlich sogar bis du das jetzt schriebst nicht - ich hab einfach nur den Code um das ganze Gerümpel bereinigt und dabei auch die digitalWrites auf 13 rausgeworfen. Eine Verbindung habe ich natürlich nicht hergestellt, logisch - ich Depp!

Danke aber trotz für's auf die Sprünge helfen! Das passiert mir auch nicht wieder... uiuiui...

MicroBahner, 2024-02-15:
Hi Holger,
wie sagt der Engländer 'shit happens'. Das ist nunmal so, und da sind wir alle nicht vor gefeit.
Jedenfalls super, dass es jetzt funktioniert.
Franz-Peter