MicroBahner/MobaTools #26: Control the stepper enable pin manually in the code rather than relying on a delay

rafik24, 2024-02-16:
Hi MicroBahner,

First of all, Thank you for this great library !

In my application the stepper motors have some weight on when moving, so I need to control the enable pin to ensure they stay where they are at certain times and also that they are disabled by the code when needed.

I looked at the documentation and could not find a way to do this other than setting a delay in stepper.attachEnable().

I there a way to do this or will it require implementation ?

Kind regards,
Rafik

MicroBahner, 2024-02-17:
Hi Rafik,
as it is now, if you use .attachEnable() the motor is always disabled if it is not moving. Only the delaytime between the last step and disabling ( and between enabling and the first step ) can be set. If you need to disable the motor only sometimes, and sometimes not when it is not moving, you cannot use attachEnable(), but must controll the enable pin by yourself.
But I think it would be a valuable extension, to be able to suppress the automatic disabling if needed. That should not nee too much effort, and I will put it on my ToDo list.
Regards
Franz-Peter

rafik24, 2024-02-17:
Hi Frantz,

Thank you for your reply.

Indeed not using attachenable and controling the enable pin manually in the code does makes sense for me.

I looked into making the change myself, I could trace the behaviour in the code but my skills in c++ are not there yet 😕

Thank you for adding this to your list.

Many thanks,
Rafik

rafik24, 2025-02-07:
Hey @MicroBahner,

This is great news !

I will give it a drive ;-)

Many thanks for the update.

Cheers,
Rafik