MicroBahner/MobaTools #27: Support for newer ESP32 C3, S3 series?

WilliamLehmus, 2024-02-17:
Hi!
Is there any possibility to see support for ESP32 C3, S3  in the future? 
If not. Do you guys have any suggestions on ESP32 based boards that work? 

BR
William

MikeyMoMo, 2024-02-20:
I just tried the buttons routines on an S3 and got lots of errors with the sample program 
"ButtonTest" you provided.  I would surely like to see this work, there.  button2 is giving me fits and something else to try would be nice.  Thanks!  I know how much trouble it is to distribute and support free code.  I have done it for years.  Your efforts are appreciated.

```
In file included from D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:5:
D:\Arduino\libraries\MobaTools\src\esp32\drivers.h:7:3: error: #error This ESP32 version is not supported
  #error This ESP32 version is not supported
   ^~~~~
In file included from C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_reg.h:21,
                 from D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:107:
D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c: In function '_initLedcHw':
D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:166:34: error: 'DPORT_PERIP_CLK_EN_REG' undeclared (first use in this function); did you mean 'SYSTEM_PERIP_CLK_EN1_REG'?
     if(!(DPORT_GET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_LEDC_CLK_EN)) ||
                                  ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:74:61: note: in definition of macro '_DPORT_READ_PERI_REG'
 #define _DPORT_READ_PERI_REG(addr) (*((volatile uint32_t *)(addr)))
                                                             ^~~~
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:91:47: note: in expansion of macro 'DPORT_READ_PERI_REG'
 #define DPORT_GET_PERI_REG_MASK(reg, mask)   (DPORT_READ_PERI_REG(reg) & (mask))
                                               ^~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:166:10: note: in expansion of macro 'DPORT_GET_PERI_REG_MASK'
     if(!(DPORT_GET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_LEDC_CLK_EN)) ||
          ^~~~~~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:166:34: note: each undeclared identifier is reported only once for each function it appears in
     if(!(DPORT_GET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_LEDC_CLK_EN)) ||
                                  ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:74:61: note: in definition of macro '_DPORT_READ_PERI_REG'
 #define _DPORT_READ_PERI_REG(addr) (*((volatile uint32_t *)(addr)))
                                                             ^~~~
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:91:47: note: in expansion of macro 'DPORT_READ_PERI_REG'
 #define DPORT_GET_PERI_REG_MASK(reg, mask)   (DPORT_READ_PERI_REG(reg) & (mask))
                                               ^~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:166:10: note: in expansion of macro 'DPORT_GET_PERI_REG_MASK'
     if(!(DPORT_GET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_LEDC_CLK_EN)) ||
          ^~~~~~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:166:58: error: 'DPORT_LEDC_CLK_EN' undeclared (first use in this function); did you mean 'LEDC_CLK_EN'?
     if(!(DPORT_GET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_LEDC_CLK_EN)) ||
                                                          ^~~~~~~~~~~~~~~~~
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:91:75: note: in definition of macro 'DPORT_GET_PERI_REG_MASK'
 #define DPORT_GET_PERI_REG_MASK(reg, mask)   (DPORT_READ_PERI_REG(reg) & (mask))
                                                                           ^~~~
D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:167:33: error: 'DPORT_PERIP_RST_EN_REG' undeclared (first use in this function); did you mean 'SYSTEM_PERIP_RST_EN1_REG'?
        (DPORT_GET_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_LEDC_RST) ) )  {
                                 ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:74:61: note: in definition of macro '_DPORT_READ_PERI_REG'
 #define _DPORT_READ_PERI_REG(addr) (*((volatile uint32_t *)(addr)))
                                                             ^~~~
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:91:47: note: in expansion of macro 'DPORT_READ_PERI_REG'
 #define DPORT_GET_PERI_REG_MASK(reg, mask)   (DPORT_READ_PERI_REG(reg) & (mask))
                                               ^~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:167:9: note: in expansion of macro 'DPORT_GET_PERI_REG_MASK'
        (DPORT_GET_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_LEDC_RST) ) )  {
         ^~~~~~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:167:57: error: 'DPORT_LEDC_RST' undeclared (first use in this function); did you mean 'DPORT_REG_WRITE'?
[... cut ...]
     pinMatrixOutAttach(pin, (group?LEDC_LS_SIG_OUT0_IDX:LEDC_HS_SIG_OUT0_IDX) + channel, false, false);
                                                         ^~~~~~~~~~~~~~~~~~~~
                                                         LEDC_LS_SIG_OUT0_IDX
D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c: In function 'setPwmDuty':
D:\Arduino\libraries\MobaTools\src\esp32\driversESP32.c:256:44: error: 'volatile union <anonymous>' has no member named 'clk_en'
             LEDC_CHAN(group, channel).conf0.clk_en = 0;

```

MicroBahner, 2024-02-20:
Yes, the S3 is stilll not supported. MoToButtons should work however, because there are no HW dependent calls. But because the IDE tries to compile everything, you get the errors.
If you want to use MoToButtons on an S3, it should be possible to copy the MoToButtons.h into your sketch directory and include with 
`#include "MoToButtons.h"`
It does not need anything from the rest of the MobaTools library.

lee32768, 2024-04-23:
Is it possible to do that with timers for the ESP32S3 as well?

MicroBahner, 2024-04-24:
What do you want to do with ESP32S3 timers?

lee32768, 2024-04-25:
Your timer code  - where you give it a setpoint to start the timer, and then call a function to see if the timer has expired - it's very intuitive to how I perceive timers from the industrial control world.  I wrote some code with your library and hit compile before I realized my target, the S3, was not supported.

MicroBahner, 2024-04-26:
Yes, that's not the problem and will be the same as with ESP32 when I support the ESP32S3.
The problem with ESP32S3 is the LED-PWM HW which I use for Servos and bulb simulation. This HW is restricted on the ESP32S3 compared to the ESP32. ( only 8 PWM channels compared to 16 on ESP32 ). That's a  bit few. Maybe I will control the bulb simulation also in the timer interupt as with the other platforms. Or I use the MC-PWM Hw also. I'm not yet sure about this.