MicroBahner/MobaTools #28: Stepper does not do exact degrees

Al-Vista, 2024-02-19:
- I am using Arduino Uno R4 Wi-Fi, connected via TMC2208 to a 200-step Nema17 motor. The application is a telecine, I need to advance the film exactly one frame per exposure of the camera, so I tried to use the _myStepper.write(30)_ command to rotate the sprocket (which has 12 teeth) by one frame. However it actually loses about 1° per 360° of rotation, so after a few minutes the frames will be completely out of sync. 
- I set the steps per rev to 1600 as that is what gives approximately 30° even though the nominal step count of the motor is 200, perhaps the driver is doing micro stepping by itself.
- I have tried reducing the speed in case it is missing steps, but that seems to make no difference. How should I be doing this?
[image]

MicroBahner, 2024-02-20:
> I set the steps per rev to 1600 as that is what gives approximately 30°

This 'approximately' is your problem. Your stepper cannot do exactly 30 degrees. MobaTools tries to adjust as close as possible, but there will be a difference. This is usually compensated when the target angle is increased in 30° steps, for example: 30 ... 60 .. 90 .. 120 .. so the error will not accumulate. But your setZero() with every increment inhibits that.
Therefor define a target position, e.g. 
`long targetAngle;`
and incremnt this with every iteration and don't use a setZero() command:
```
myStepper.write(targetAngle);
targetAngle += 30;
```

Al-Vista, 2024-02-20:
Thanks! I'll try that.

MicroBahner, 2024-06-05:
.

MicroBahner, 2024-06-05:
.

closed 2024-06-05 by MicroBahner