MicroBahner/MobaTools #31: Question about timers and functionality

zhiganik, 2024-03-10:
**Hi, first of all I would like to say thank you for the awesome library!**

My question is that I want to make a program that manages tasks.
One task contains one or more motors. Each motor has a different speed and number of steps they have to take.
There can be many such tasks (20-30) and the maximum number of motors in a task can be 6.
When the task is executed, the motors that are set in the task must pass their set steps with their own speed, when all the motors in the task have completed his work - it is necessary to switch to the next task.
Is it possible to perform this behavior with _MobaTools_ and will 6 motors start working at the same time without losing steps?

Is it possible to use more than 6 MoToSteppers (not at the same time)?

**My Setup:**
- ESP32 WiFi + BT 4.2 | ESP32-DevKit
- 6x NEMA 23 2.8A 1.89Nm
- 6x 5.6A DC 24-50V 57/86 2 Phase Stepping Driver DM556
- ESP32 is powered with 5v

I would also like to ask about timers working with _MoToStepper_. This my test code does not work (the motor spins endlessly, and if you output timer.getRemain() in loop method, it will output the time once or several times and that's it, while in _SerialMonitor_ there are no errors), could you help ?

```
#include <MobaTools.h>

#define STEPS_PER_REVOLUTION 32
#define PUL_PIN_1 16
#define DIR_PIN_1 17

const int stepsPerRevolution = 200; 

MoToStepper Stepper1(stepsPerRevolution * STEPS_PER_REVOLUTION, STEPDIR);
MoToTimer timer;

void setup() 
{
  Serial.begin(9600);
  Stepper1.attach(PUL_PIN_1, DIR_PIN_1);

  timer.setTime((long)400);
  Stepper1.rotate(1);
}

void loop() 
{
  if(!timer.running())
  {
    Stepper1.stop();
  }
}
```

MicroBahner, 2024-03-10:
Hi,
> There can be many such tasks (20-30) and the maximum number of motors in a task can be 6.

Because HW (timer) is used, the maximum number of motors cannot be increased by using many tasks. The ESP32 may be able to control more than 6 motors, but I never tested that. It's surely also dependent on the speed of the steppers. All steps are created within a timer ISR.
The max number of motors is not limited by the SW structure. There is a #define in MobaTools.h the sets the limit:
`#define MAX_STEPPER     6       // 
`
You can try to increase this.

> This my test code does not work (the motor spins endlessly, and if you output timer.getRemain() in loop method, it will output the time once or several times and that's it, 

If I use your code it works. Here's a [WOKWI siumulation](https://wokwi.com/projects/392002758973376513)

I would recommend to use the expired() method to stop the stepper.

> will 6 motors start working at the same time without losing steps?

Depends on your HW. The MoToStepper class can create all the steps at the same time.

zhiganik, 2024-03-11:
**Hi, thank you for your reply!**

- I will test in the future a platform with 8 motors simultaneously. 
- Yesterday I managed to make a sketch for simultaneous operation of 6 motors, for each of them I pass in `doSteps(800 * 32)`, and in `setSpeedSteps(800 * 32 * 10)` - which assumes that all motors will pass full 4 revolutions with set microstep 1/32 (6400) in 1 second.

Everything works fine, no problems with loss of steps at this speed, but in reality the motors pass these steps in 1.2-1.3 seconds, how can I get exactly 1 second? Maybe I don't understand how `setSpeedSteps`.... works or is this error due to high speeds and number of steps ?

Regarding the timed stop problem, I'll check and see why it doesn't start for me soon.

MicroBahner, 2024-03-11:
> but in reality the motors pass these steps in 1.2-1.3 seconds, 

Did you define a ramp?

>  or is this error due to high speeds

Well, your fairly high step speed needs an ISR about every 20µs ( switching the pulses on and off ). If there is another ISR load on the ESP this may delay things a little ... 
Is the timing more exakt at lower speeds?

MicroBahner, 2024-06-05:
Obviously no more question

closed 2024-06-05 by MicroBahner