MicroBahner/MobaTools #33: Migration related issues Espressif ESP32 updated from 2.X to 3.0

zs6buj, 2024-06-04:
Thank you for your awesome library.

Arduino Espressif ESP32 recently updated from 2.X to 3.0, and several timer functions have been simplified. As a result the MobaTools for ESP32 fails to compile. I patched several timer functions and the Servo01 example now compiles, but problems might arise elsewhere. My testing was done using an DOIT ESP32 DEVKIT V1.

The new timer functions are documented [here](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/timer.html)

All 3.0.0 version migration related issues are discussed [here](https://github.com/espressif/arduino-esp32/issues/8796) if you would like to look at them.

I could do a PR, but I suspect there will be implications that I might miss.

Thank again for this amazing library.

MicroBahner, 2024-06-05:
Thanks for the hint and the links. I will look at that.

zs6buj, 2024-08-21:
New update of Moba_Tools is working nicely with ESP32 Core 3.x.x now. Thank you. I shall close the issue.

closed 2024-08-21 by zs6buj