MicroBahner/MobaTools #36: No inrease of brightness in MoToSoftLed

hatjan, 2024-08-21:
German version below

Hello Franz-Peter,
First of all: I am very happy to have found your MoBaTools. My current project is a ceiling lamp with two LEDs that I can not only switch on and off and dim, but can also move up and down about a meter with three stepper motors. I was able to use almost all of the MoBaTools classes for this.
My problem: I cannot properly increase the brightness of an LED that is already switched on (lets say from 10% brightness to 60%). From your code I have seen that on(bri) only works when the LED is currently off.
I tried to get around the problem by first switching the LED off - but with the current brightness as parameter - and then switching it back on with the desired value. This works in principle, but the new brightness is set immediately - no dimming to that point. In this case, dimming only takes place if I wait about 3 seconds after temporarily switching it off.
Is there a solution to this problem that I have overlooked?
Apart from that, it would be helpful if one could query the current brightness while dimming, similar to stepsToDo() in MoToStepper.
Thank you very much.

Hallo Franz-Peter,
zunächst einmal: Ich bin sehr glücklich deine MoBaTools gefunden zu haben. Mein aktuelles Projekt ist eine Deckenlampe mit zwei Leds, die ich nicht an- und ausschalten sowie dimmen kann, sondern auch mit drei Schrittmotoren ca. einen Meter rauf- und runterfahren kann. Hierfür konnte ich fast alle Klassen der MoBaTools verwenden.
Mein Problem: Es gelingt mir nicht sauber, die Helligkeit einer bereits eingeschaltten Led (10% Brightness) zu erhöhen (60%). Aus dem Code habe ich erkannt, das on(bri) nur funtioniert, wenn die Led gerode off ist.
Ich habe versucht das Problem zum umgehen, indem ich die Led zunächst - mit der aktuellen Helligkeit als Parameter -  ausschalte und dann mit dem gewünschten Wert wieder einschalte. Das funktioniert prinzipiell, aber die neue Helligkeit wird sofort eingestellt - kein dimmen dorthin. Gedimmt wird in diesem Fall nur, wenn ich nachdem temporären Ausschalten knapp 3 Sekunden warte.
Gibt es eine Lösung für dieses Problem, die ich übersehen habe?
Davon abgesehen, wäre es hilfreich, wenn mann während des Dimmens die aktuelle Helligkeit abfragen könnte, ähnlich stepsToDo() in MoToStepper.
Vielen Dank.

MicroBahner, 2024-08-24:
Hallo hatjan,
Es freut mich, wenn Du die MobaTools nutzen kannst.
Welchen Arduino verwendest Du? nicht alle unterstützen das auf/abdimmen mit Zwischenzuständen. Ursprünglich war das auch gar nicht zum beliebiegen 'Dimmen' einer Led gedacht, sondern nur zum weichen Auf/Abblenden (Glühbirnensimulatione). Denn die PWM wird komplett per SW erzeugt, und da sollte der Prozessor nicht ständig belastet sein. Bei den schnelleren Prozessoren habe ich zugelassen, dass die Endzustände auch nicht vollständig AUS bzw EIN sein dürfen. Aber als komplett freier Dimmer mit beliebigem heller/dunkler setzen war das nie gedacht, und so habe ich es auch nie getestet. Vermultich liegt da das Problem.
Franz-Peter

hatjan, 2024-08-24:
Hallo nochmal,
vielen Dank für die Antwort. Ich benutze ein ESP 32 DevKit, an der Hardware wird es wohl nicht liegen, eher, wie du beschrieben hast, am eigentlichen Zweck der Klasse.
Ich habe mir für die Leds inzwischen eine andere Library (FastLed, ESP32-Fork) heruntergeladen und die passt für mich perfekt.
MoToStepper und MoToTimer benutze ich weiterhin sehr gern kann ich nur empfehlen.
Gruß aus Velbert, NRW

MicroBahner, 2024-09-05:
Closed the issue , because MoToSoftled isn't intended to be used that way.

closed 2024-09-05 by MicroBahner