MicroBahner/MobaTools #38: Open-drain GPIO in MotoStepper

kf1375, 2024-10-18:
Hi @MicroBahner,

Since many stepper motor drivers must be controlled by Open Drain output, it would be great to support this. Currently, I achieved this by changing some lines in the "src\utilities\MoToStepper.cpp" and "src\utilities\MoToStepperNo8266.inc" but it would be great if this existed in the library.

Thanks

MicroBahner, 2024-10-22:
Hi kf375,
Can you give an example of such a stepper driver? So far I have never seen a stepper driver that requires an OpenDrain output at the Arduino.

kf1375, 2024-10-24:
Thanks for your reply,

[image]
 
https://www.omc-stepperonline.com/digital-stepper-drive-1-0-3-2a-18-30vdc-for-nema-17-23-stepper-motor-dm332t

As an example this driver motor requires an OpenDrain output.

MicroBahner, 2024-10-24:
It doesn't **need** an OpenDrain output. If you use Vcc=5V, you can safely connect PUL, DIR and ENA directly to an Arduino OUTPUT. There are a lot of drivers with such an optoisolated input. All work well together with MoToStepper.

kf1375, 2024-10-27:
As well as I am using ESP32, the Vcc would be 3V3. Both when the pin is low and when the pin is high, the current passes through the input diode of the optocoupler. Is it not safer to drive the pins in OpenDrain mode? It will ensure that the optocouplers will completely turn off and turn on.

MicroBahner, 2024-10-28:
Vcc=3.3V is out of spec for the inputs of the DM332T. So with an ESP32 you need an external driver for reliable operation.
If the output pin is HIGH ( within spec, Vcc=5V ) no current will flow through the optocoupler.
Even if you create an opendrain output at the Arduino, Vcc for the optocoupler must not be higher than Vcc of the Arduino. So opendrain or push-pull makes no difference when connecting the optocoupler inputs directly to the Arduino.

MicroBahner, 2024-10-28:
From the datasheet:

> Pulse signal: Pulse active at rising edge; **4-5V** when PUL-HIGH, **0-0.5V** when PUL-LOW.

kf1375, 2024-11-18:
Thank you, @MicroBahner, for the detailed clarification. We've decided to use an external driver in our next update based on your advice. Much appreciated! 🙏