MicroBahner/MobaTools #43: Questions about swapping stepper rotation direction directly after rotating after completing rotating in one direction

jp660690, 2025-01-27:
Good Day MicroBahner,
Thank you for creating this library.  It has been very helpful in getting my project off the ground.  I do have one question about the MoToStepper portion of the library.  I have the stepper rotating a number of steps in one direction and once complete, I need it to rotate in the opposite direction immediately after in the other direction. As my code is written, I have it move a the required number of steps then move the opposite direction a specific number of steps.  As of this time, it only moves in the initial direction, but it does not move in the opposite direction once complete.  I have attempted to insert an interrupt time delay between the two lines of code, but it still does not work.  My question is, Do I need to have a non-interrupt type of delay for this to work?   I have included the snippet of code here;

 myStepper.move(3360);
    delay(1000);
 myStepper.move(-2000);

Thanks ahead of time for your assistance.

jp660690, 2025-01-28:
Thanks for answering my question.
Last night while digging deeper, I found one of your examples (back_ForthStepperPause.ino) that used an if/else construct with a delay and when I integrated this into my code, it did not initially work due to the pause timing. but after adjusting it, I got it to work perfectly.  This movement action now works properly and correctly. (see code below)

  //Dispense button action
  if(button.pressed(Button3))
  {
    myStepper.doSteps(dispenseSteps);
    Serial.println("Dispense");
    stepperRunning = true;
  }
  if(stepperRunning)
  {
    if(!myStepper.moving())
    {
      stepperPause.setTime(500);
      stepperRunning=false;
    }
  }
  else
  {
    if(stepperPause.expired())
    {
      myStepper.doSteps(snuffBack);
      Serial.println("Snuff back");
    }
  }

Again, thanks for creating this library and thanks for your response to my question.

MicroBahner, 2025-01-29:
You are welcome

closed 2025-01-29 by MicroBahner