MicroBahner/MobaTools #44: Support for RP2040 / 2350

chatelao, 2025-01-29:
Q1:
- Is there any plan to support for RP2040?

Q2:
- Which files would need a patch to support RP2040s?

MicroBahner, 2025-01-29:
Yes, the next release will support the RP2040 and RP2350. I recently started the development ( but be patient, it may still take some time ). 
There will be a new set of files for the RP2040 ( the same as with the other supported processors ). I hope  I don't need to change any of the common files ( but who knows ...)

chatelao, 2025-01-30:
If you would like to try "RailCom" support too -  I have designed this OSHW breakout board:
- https://www.tindie.com/products/chatelao/dccrailcom-interface-for-arduino-decoders

And there is a turnout decoder (with RailCom) too:
- https://www.tindie.com/products/chatelao/dcc-open-source-decoder-for-4-turnouts

Just drop me a note if one or both would help you for testing. I would be more than happy to send them for free.

chatelao, 2025-02-24:
I will try to add it here:
- https://github.com/chatelao/NmraDcc/tree/railcom-lib

It is VERY early beta, not even tested just compiling.

chatelao, 2025-02-25:
Okay, thank you for the feedback. I guess you will see as soon as RailCom will work on NmraDcc, then maybe some feedback from decoders about turnout/servo positions will be possible too.

MicroBahner, 2025-03-04:
V2.7.0 supports RP2040

closed 2025-03-04 by MicroBahner