MicroBahner/MobaTools #45: Implementing "myStepper.stepsToDo()"

jp660690, 2025-02-06:
Good Day!

As I have stated previously, thank you for creating this great library.  It has shortened my development time considerably.

While looking through the README file, in the MoTo Stepper section, there is what I think is a function called "long myStepper.stepsToDo();", which would be handy for what I am trying to achieve.  I looked through your examples and could not find an example of how is this used.  
As some background, what I am trying to do is have the stepper move a specific number of steps each movement cycle.  If during a movement cycle, an emergency stop commend is received (using myStepper.stop();), I would like to know how many more steps are left in the current movement cycle.  So once the cause of the emergency stop has been cleared, I would like the stepper to finish the current movement cycle using the steps that are remaining that were retrieved through myStepper.stepsToDo().
So I have the following questions,
1. For each movement cycle, do I have to set a zero point in order to use this?  I was hoping to use relative positions instead of absolute positions.
2. I am presently using myStepper.doSteps() to initiate movement.  If I must use absolute positions, it seems that I need to use myStepper.writeSteps(long stepPos); or can I keep using myStepper.doSteps()?

I know that there is a way to count steps as the stepper rotates but this seems cumbersome to implement and is probably not necessary for this use case.

Thanks ahead of time for your answers to my questions.

MicroBahner, 2025-02-07:
Hi,

> For each movement cycle, do I have to set a zero point in order to use this? I was hoping to use relative positions instead of absolute positions.

No, using '`myStepper.stepsToDo`' is independent of absolute or relative positioning. For MoToStepper effectively there is no difference between both kinds of positioning. If using relative positioning ( `doSteps()` ) it is internally converted to a new absolute position that is 'doSteps' steps away from the actual position. MoToStepper always knows the absolute position of the stepper, regardless wich method of positioning/moving you use.
'`myStepper.stepsToDo()`' always returns the actually needed steps to reach the target - independent from how this target was defined. You can execute this command at any time.

> using myStepper.stop();

When using this, step creation is stopped immediately even if a ramp is defined. And the actual position is defined as the actual target ( NOT as zero point ). To remember the remaining steps to the previous target, you must execute `stepsToDo()` immediately before executing the `stop()` command. But be aware that your motor may not be able to stop within one step distence, depending on the speed and moving mass. It could be that the stepper no longer is positionied where MobaTools expects it to.

jp660690, 2025-02-14:
I apologize for not responding sooner to your reply.  I will keep what you have said in mind about the "myStepper.stop()" and re-write my code to eliminate this.
One further question on 'myStepper.stepsToDo()'.  How is this written in code? Unfortunately, I am relatively new to Arduino and to programming in general, but I am willing to learn.  Is this a variable that is initialized prior to the loop as 'long myStepper.stepsToDo();' and then when I wish to get this value, I will have to create another variable and have this equal to 'myStepper.stepsToDo()' or do I create a variable and then have it equal to myStepper.stepsToDo (ex. stepsRemain = myStepper.stepsToDo)?

MicroBahner, 2025-02-16:
'`myStepper.stepsToDo()`' is not a variable but a function (method) of the MoToStepper class that returns the remaining steps as it is in the moment you call the function. The return value will change with every call as the motor approaches the target.
 If you want to remember a value for later use, then you have to store it in a variable of your own.

jp660690, 2025-02-17:
If you could correct me, the code should look like this:
"stepRemain = myStepper.stepsToDo();"
with
"stepRemain" being the variable that I create that equals the function call?

As always, thanks for putting up with my questions.  I greatly appreciate it.

MicroBahner, 2025-02-17:
Yes, this saves in 'stepRemain' how many steps remain to the target at the time you execute this call.

jp660690, 2025-02-17:
Thanks for the help!!

closed 2025-02-17 by jp660690