MicroBahner/MobaTools #46: documentation of timer usage is missing

gitterman, 2025-02-17:
I found your great library but cannot find anything about the timer resources used by it.
I want to use my own ISR on an arduino to measure some input signal frequency while running my stepper motor.
For this, I need to know which timer/counter functions I can use without interfering with MobaTools.

MicroBahner, 2025-02-17:
Which Arduino are you using? Chapter 2 of the documentation describes which timers are used in the various supported processors.

gitterman, 2025-02-17:
Thanks for the quick reply.
I am using an Arduino Nano, and I am sorry that my question wasn't clear enough.
I want to know, in detail, which function of which timer is used or is not used on the AVR.
E. g. can I use timer compare A and B interrupts on timer0?
I understand that your library uses timer1.
Do you start or stop the timer or do you use it in free running mode like timer0 is used with the millis().
If so, are all vectors used by your library or are certain ISR vectors still available, like TIMER1_CAPT_vect?

MicroBahner, 2025-02-18:
> I want to know, in detail, which function of which timer is used or is not used on the AVR.
E. g. can I use timer compare A and B interrupts on timer0?

On AVR with only one 16-bit timer, MobaTools uses timer1 for various functions, and you cannot use it for any other purpose without disturbing the MobaTools functionality. There maybe even changes how this timer is used in future releases.
If you need a 16bit Timer for your purpose, you need to use an AVR with more 16-Bit timers, e.g. an Arduino Micro/Leonardo ( which has 2 16-bit timers ) or a Mega, which has even more.
On an AVR with timer3 ( which is 16-bit) this timer is used by MobaTools, so that timer1 can be freely used at such MCU's
Other timers are not affected by MobaTools. But there are MobaTools functions that use millis() - so you should not disturb the millis() function of the Arduino subsystem.

gitterman, 2025-02-18:
Ok, thank you for clarifying.
I will refrain from any usage of timer1 with mobatools on my arduino nano then.

closed 2025-02-18 by gitterman