MicroBahner/MobaTools #48: Semaphore Signal Bounce

Cobalt6700, 2025-03-25:
Hi MicroBahner,

First off - thanks for the great library - I use it on all of the controllers I have made for my garden railway.  

I have just got into doing signals - and whilst I have light signals sorted, I would like to make my semaphore signals look a bit more realistic. 

I am already using a different speed for each aspect, but I wondered if there is a way to use your library to get some sort of 'bounce' at the ends of the servo travel? There are a few examples of bounce code around, I have had a look at them but don't understand how I would get these to work with MobaTools. 

Any advice would be greatly appreciated. 

Thanks

MicroBahner, 2025-03-26:
The library itself doesn't support 'bouncing' of servos. But of course you can do it in your sketch. But I didn't do so with my signals, because I was very disapointed about the result. Because of the gear lash in the servo it doesn't look realisitic.

closed 2025-03-26 by MicroBahner

Cobalt6700, 2025-03-26:
Thanks for the quick reply. 

I will have a look at doing that - is there a limit to how quickly you can call the servo.write() function in the loop?

MicroBahner, 2025-03-27:
> is there a limit to how quickly you can call the servo.write() function in the loop?

I'm not shure what is 'quickly' in your opinion. But there is shurely no problem to call it at the bouncing-points. You can in any case  call servo.write() quicker than your semaphore arm can change its direction when bouncing ;-) .

Many years ago ;-) , I programmed the bouncing of a railroad barrier. But that was with a small stepper motor and a rubber band to preload the gear so that there was no gear backlash.  [Video](https://youtu.be/i2poWsW97s4)
The programming of the bouncing should be similar to your semaphore signals. I can look if I still have the source code, if you want. But it was definitely commented in german.

Cobalt6700, 2025-03-27:
Some information from me on the speed would have been useful :) 
The sketch I was having a go with has a cycle/loop time of 5000us. If the sketch is actually running at 200Hz is something I need to confirm on the oscilloscope. 

Nice barrier. That's some setup under the baseboard! 
I would certainly be interested to have a look yes - however if it is buried somewhere then don't go to too much trouble. 

No problems with German - I know a little, and I used a VS Code plugin to translate your DCC Zubehoerdecoder library so I could find out how it worked and start modifying it to run on an ATtiny 3226 :)

MicroBahner, 2025-03-27:
This is the Sketch for the complete railroad-crossing: [BueSteuerung_V0.5S.zip](https://github.com/user-attachments/files/19493778/BueSteuerung_V0.5S.zip) . The moving of the stepper(s) is done in the function `Schrankenbewegung` at the end of the sketch.

> ... DCC Zubehoerdecoder library so I could find out how it worked and start modifying it to run on an ATtiny 3226 :)

Wow - impressive project.

Cobalt6700, 2025-03-27:
Thanks very much, I will have a look at the code and see what I can get from it :)

I would say I'm nearly there. Got it up and running on a breadboard, light signals and servos working well.

Had to modify the MobaTools timer configuration to get it working, I may well open a pull request sometime soon - though I have only tested the softled and servo functions.