MicroBahner/MobaTools #49: Adding support for Attiny 2 series

Cobalt6700, 2025-04-06:
Hi MicroBahner,

I'm working on adding support for the Attiny 2 series. I think I'm getting there and think I have managed to setup TCA0 with the correct prescaler for 0.5uS/tic. 

I see that you have a number of debug pins setup, for example, SET_TP2, CLR_TP4: 
```
ISR (TCA0_CMP0_vect) {
     TCA0.SINGLE.INTFLAGS = TCA_SINGLE_CMP0_bm;	// Reset IRQ-flag
     #warning motoservo megatinycore
    SET_TP2;
    CLR_TP4;
    // Timer1 Compare A, used for servo motor
    if ( IrqType == POFF ) { // Pulse OFF time
....
```
Could please you advise which test pins and the method you used to confirm that you are doing 0.5us/tic? I will then be able to confirm that the changes have worked :)

Thanks

I have included the timer code for the attiny 2 series below:

```
void seizeTimerAS() {
    static bool timerInitialized = false;
    if ( !timerInitialized ) {
        // using timer TCA0 in normal mode.
        // CMP0 register used for steppers and softleds
        // CMP1 register used for servos
		// CMP2	register (not yet )used for  softleds
        noInterrupts();
        takeOverTCA0();
        // TCA0.SINGLE.CTRLESET = TCA_SINGLE_CMD_RESET_gc;     // hard reset timer // Not required because of takeOverTCA0()
        // For ATtiny with megaTinyCore, we can use DIV8 prescaler since millis() uses TCB1
        TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL_DIV8_gc;      // 0.5µs per tic with 16MHz clock
        // On standard megaAVR we would need to use DIV64: TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL_DIV64_gc;
        // megaTinyCore allows a prescaler of 2 for TCB1 for millis(), so changing TCA0 doesn't influence millis()
        TCA0.SINGLE.CTRLB = TCA_SINGLE_WGMODE_NORMAL_gc;    // normal mode, no autolockUpdate, no pins active
        TCA0.SINGLE.CTRLC = 0;
        TCA0.SINGLE.CTRLD = 0;                              // no split mode ( user 16bit timer )
        //TCA0.SINGLE.CTRLECLR                              // wasn't used in megaavr code - not sure what it would do here
        TCA0.SINGLE.CTRLESET = TCA_SINGLE_LUPD_bm;          // don't use buffered compare registes
        //TCA0.SINGLE.CTRLFCLR                              / wasn't used in megaavr code - not sure what it would do here
        //TCA0.SINGLE.CTRLFSET                              // wasn't used in megaavr code - not sure what it would do here
        TCA0.SINGLE.INTCTRL = TCA_SINGLE_CMP0_bm | TCA_SINGLE_CMP1_bm; // enable cmp0 and cmp1 interrupt
        TCA0.SINGLE.INTFLAGS = 0;   // clear all interrupts
        TCA0.SINGLE.PER  = TIMERPERIODE * TICS_PER_MICROSECOND;  // timer periode is 20000us
        TCA0.SINGLE.CMP0 = FIRST_PULSE;
        TCA0.SINGLE.CMP1 = 400;
        TCA0.SINGLE.CTRLA |= TCA_SINGLE_ENABLE_bm;          // Enable the timer
        interrupts();
        timerInitialized = true;
        MODE_TP1;   // set debug-pins to Output
        MODE_TP2;
        MODE_TP3;
        MODE_TP4;
        DB_PRINT("CYCLETICS=%d, TIMER_OVL_TICS=%d", CYCLETICS, TIMER_OVL_TICS );
    }
}
```

MicroBahner, 2025-04-07:
Hi,
interesting ... I never had a look at the ATTiny 2 series so far. Are there any boards already, or do you have to use the bare chip?

The test pins are defined board specific in  src/utilities/MoToDbg.h

Cobalt6700, 2025-04-08:
I mainly moved to them because the new ATTiny chips need very little external components (2 caps), upload using UPDI (3 wires) and even come in a 4mm x 4mm footprint - great for making small accessory decoders. 

Dr Azzy (Spence Konde, who built megatinycore) did have some boards available, however they are currently out of production: [](https://www.tindie.com/products/drazzy/attiny322632161606-dev-board-arduino-compatible/)

Currently I am using the SOIC 20 chips on breakout boards, but I am in the process of getting some PCB's made - once I have confirmed I have the code working properly :) . This is with the intention of making a 6 channel / 12 output decoder that measures 22x40mm. 

Thanks for confirming where the pins are defined - can you help me with what the output from these pins should look like? I will connect it to my oscilloscope and try to understand, I thought it would be good to ask rather than guessing. 

Thanks

MicroBahner, 2025-04-09:
Well the meaning of the testpins isn't generally fixed. I usually change the position of SET_TPx and CLR_TPx relating to the test I do at that moment. So what you can see in the code is fairly random relating the last tests I did. I use it e.g. for checking how long an ISR lasts, when it is called or to test if a part of a code is executed when it should (or shouldn't). Its intended for real-time testing, where Serial prints are not possible ( especially within interrupts). Before releasing a version I switch them all off in MoToDbg.h so it doesn't matter where they are at that moment. 
So feel free to move that testpoints around as you need for your tests.

PR #50 (2025-04-18): Adding ATtiny 2 series support
Resolves #49 

These updates have been tested on an ATtiny 3226 and 3227 using megaTinyCore. It is likely that other 2 series parts with enough Flash + RAM would also work. 

I have tested that the library functions; button, timer, servo, softled. I have not tested stepper.